ros-wg-delib/examples #48: Pyrobosim UI gets out of sync with simulation state

dcconner, 2024-09-25:
I’m testing with a local build on Iron, but I repeatedly have issues with PyRoboSim UI getting out of sync and not updating.

I have not discerned a repeatable pattern to when it gets out of sync, but it is likely during path following.
It will often run repeated commands without issue before getting out of sync.

The robot_state is correct so it’s just the UI view.  If I activate a button on UI then it normally updates to current state, but it seems like once the UI gets out of sync it stays out of sync until I manually click some button on the UI.

Has anyone else seen anything like this?

sea-bass, 2024-09-30:
We debugged this and made a handful of fixes to PyRoboSim, which are on main here now.

The problem doesn't seem to be gone entirely, but it has been reduced.

dcconner, 2024-10-03:
I'm happy with performance with recent fixes.

closed 2024-10-03 by dcconner