ros-wg-delib/examples #52: Fails to find Package 'delib_ws_p1' while running setup instructions for ROSCon 2024 workshop on Deliberation Technologies

srikant-v, 2024-10-03:
Hello,

I'm trying out the setup instructions in advance of the ROSCon 2024 workshop on Deliberation technologies, because I wanted to see if could run it in a Docker container on my Macbook.

However, before I could fully test it out, I found that the 4th step in the Setup instructions (https://github.com/ros-wg-delib/roscon24-workshop/blob/main/README.md) fails with the following error: 

`root@docker-desktop:/delib_ws# ros2 run delib_ws_p1 run`
`Package 'delib_ws_p1' not found`

I suspect that the package has been renamed. 
Could you take a look and update the instructions?
 
Thanks.

sea-bass, 2024-10-04:
Hi @srikant-v -- thanks for testing things early! We are still actively working on this repo and those instructions are not up to date at the moment.

Right now, you can run this command instead:

```
ros2 run delib_ws_worlds run
```

And if you want to change the world number (from 1 to 4)

```
ros2 run delib_ws_worlds run --ros-args -p problem_number:=1
```

ct2034, 2024-10-07:
I consider this completed. Feel free to open another issue if something else comes up.

PR #53 (2024-10-07): Update READMEs with more information
This PR updates the READMEs with more detailed information about the setup, dev environment, and technologies we are showing the workshop. It also rearranges things into the subfolder READMEs as appropriate.

I also updated the Markdown lint to a tool that runs in pre-commit, so there is only less CI job, and added a `media` subfolder to place images into.

I would suggest keeping the review here on high-level structure, and biasing towards getting something merged. Then, everyone can adding their own extra information in the right places in follow-up PRs.
 [... cut]
closed 2024-10-07 by ct2034