ros-wg-delib/examples #68: Package Failed   <<< skiros2_task

carlos-argueta, 2024-10-14:
When I build the workspace the package skiros2_task fails. Here is the entire output:

root@ubuntu-dev:/delib_ws# delib_build
Starting >>> flexbe_msgs
Starting >>> pyrobosim_msgs
Starting >>> skiros2_msgs
Starting >>> rosbridge_test_msgs
Starting >>> rosapi_msgs
Starting >>> rosbridge_msgs
Starting >>> btcpp_ros2_interfaces
Starting >>> btlib
Starting >>> ros_bt_py_interfaces
Starting >>> pyrobosim
Finished <<< rosbridge_msgs [0.59s]                                                                                                                                                                                                                          
Finished <<< btcpp_ros2_interfaces [0.61s]                                                                                                                                                                                                               
Starting >>> behaviortree_ros2
Finished <<< behaviortree_ros2 [0.19s]                                                                                                                                                                                                                           
Starting >>> btcpp_ros2_samples
Finished <<< btcpp_ros2_samples [0.17s]                                                                                                                                                                                                                    
Finished <<< pyrobosim_msgs [1.00s]
Starting >>> pyrobosim_ros
Starting >>> delib_ws_problem_interface                                                                                                                                                                                                                          
Starting >>> pyrobosim_btcpp
Finished <<< skiros2_msgs [1.15s]                                                                                                                                                                                                              
Finished <<< flexbe_msgs [1.16s]
Starting >>> flexbe_core
Starting >>> skiros2_common
Finished <<< rosbridge_test_msgs [1.30s]                                                                                                                                                                                                             
Finished <<< rosapi_msgs [1.31s]
Starting >>> rosbridge_library                                                                                                                                                                                                                                 
Finished <<< pyrobosim [1.43s]
Finished <<< btlib [1.44s]
Starting >>> bt_view
Finished <<< pyrobosim_btcpp [0.69s]                                                                                                                                                                                                                           
Finished <<< pyrobosim_ros [0.71s]                                                                                                                                                                                                         
Finished <<< rosbridge_library [0.35s]
Starting >>> rosapi
Finished <<< delib_ws_problem_interface [1.05s]                                                                                                                                                  
Starting >>> delib_ws_worlds
Starting >>> delib_ws_python
Finished <<< flexbe_core [1.16s]                                                                                                                                                                               
Finished <<< skiros2_common [1.16s]
Starting >>> skiros2_world_model
Starting >>> flexbe_testing
Starting >>> flexbe_mirror
Starting >>> flexbe_input
Starting >>> pyrobosim_flexbe_utilities
Starting >>> flexbe_webui
Finished <<< bt_view [0.90s]
Starting >>> bt_live
Finished <<< ros_bt_py_interfaces [3.12s]                                                                                                                                                                                                                       
Finished <<< rosapi [1.35s]
Starting >>> rosbridge_server
Finished <<< flexbe_testing [0.89s]                                                                                                                                                                                                                           
Starting >>> flexbe_states
Finished <<< delib_ws_python [1.46s]                                                                                                                                                                                                           
Finished <<< rosbridge_server [0.40s]
Starting >>> rosbridge_suite
Finished <<< rosbridge_suite [0.10s]                                                                                                                                                                                                        
Starting >>> ros_bt_py_web_gui
Finished <<< delib_ws_worlds [1.65s]
Finished <<< ros_bt_py_web_gui [0.12s]                                                                                                                                                                                                          
Starting >>> ros_bt_py
Finished <<< flexbe_webui [1.80s]                                                                                                                                                                                                 
Finished <<< skiros2_world_model [1.82s]
Finished <<< bt_live [1.80s]
Finished <<< pyrobosim_flexbe_utilities [1.81s]
Finished <<< flexbe_input [1.82s]
Finished <<< flexbe_mirror [1.82s]
Starting >>> skiros2_skill
[... 39 lines cut ...]
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['/usr/bin/python3', '-c', 'import sys;from contextlib import suppress;exec("with suppress(ImportError):    from setuptools.extern.packaging.specifiers    import SpecifierSet");exec("with suppress(ImportError):    from packaging.specifiers import SpecifierSet");from distutils.core import run_setup;dist = run_setup(    \'setup.py\', script_args=(\'--dry-run\',), stop_after=\'config\');skip_keys = (\'cmdclass\', \'distclass\', \'ext_modules\', \'metadata\');data = {    key: value for key, value in dist.__dict__.items()     if (        not key.startswith(\'_\') and         not callable(value) and         key not in skip_keys and         key not in dist.display_option_names    )};data[\'metadata\'] = {    k: v for k, v in dist.metadata.__dict__.items()     if k not in (\'license_files\', \'provides_extras\')};sys.stdout.buffer.write(repr(data).encode(\'utf-8\'))']' returned non-zero exit status 1.
---
Failed   <<< skiros2_task [0.39s, exited with code 1]
Finished <<< flexbe_onboard [1.03s]
Starting >>> flexbe_widget
Finished <<< ros_bt_py [1.76s]
Finished <<< flexbe_widget [0.33s]                                                                                 
Starting >>> flexbe_behavior_engine
Finished <<< flexbe_behavior_engine [0.09s]                                                                                 
Starting >>> pyrobosim_flexbe_behaviors
Finished <<< skiros2 [1.06s]                                                                                                  
Starting >>> skiros2_std_reasoners
Starting >>> skiros2_std_skills
Finished <<< skiros2_gui [1.07s]
Finished <<< pyrobosim_flexbe_behaviors [0.59s]                                                                                                      
Finished <<< skiros2_std_reasoners [0.91s]                                                               
Finished <<< skiros2_std_skills [0.93s]
Starting >>> skiros2_pyrobosim_lib
Finished <<< skiros2_pyrobosim_lib [0.32s]                               

Summary: 46 packages finished [7.67s]
  1 package failed: skiros2_task
  1 package had stderr output: skiros2_task
Command '['/usr/bin/python3', '-c', 'import sys;from contextlib import suppress;exec("with suppress(ImportError):    from setuptools.extern.packaging.specifiers    import SpecifierSet");exec("with suppress(ImportError):    from packaging.specifiers import SpecifierSet");from distutils.core import run [... cut ...] lass\', \'distclass\', \'ext_modules\', \'metadata\');data = {    key: value for key, value in dist.__dict__.items()     if (        not key.startswith(\'_\') and         not callable(value) and         key not in skip_keys and         key not in dist.display_option_names    )};data[\'metadata\'] = {    k: v for k, v in dist.metadata.__dict__.items()     if k not in (\'license_files\', \'provides_extras\')};sys.stdout.buffer.write(repr(data).encode(\'utf-8\'))']' returned non-zero exit status 1.

sea-bass, 2024-10-14:
This should have been resolved by recent updates to SkiROS2. Could you do

```
git submodule update --init
```

And build again?

carlos-argueta, 2024-10-15:
Yeah that worked, thanks!

carlos@ubuntu-dev:~/Documents/roscon24-workshop$ docker compose run base
Starting >>> flexbe_msgs
Starting >>> pyrobosim_msgs
Starting >>> skiros2_msgs
Starting >>> rosbridge_test_msgs
Starting >>> rosapi_msgs
Starting >>> btcpp_ros2_interfaces
Starting >>> rosbridge_msgs
Starting >>> btlib
Starting >>> ros_bt_py_interfaces
Starting >>> pyrobosim
Starting >>> delib_ws_24_interfaces
Finished <<< btlib [0.85s]                                                                                                                                                                                                                                                  
Starting >>> bt_view
Finished <<< pyrobosim [0.99s]                                                                                                                                                                                                                               
Finished <<< bt_view [0.75s]                                                                                                                                                                                                                                 
Starting >>> bt_live
Finished <<< bt_live [0.75s]                                                                                                                                                                                                                                                
Starting >>> bt_tools
Finished <<< bt_tools [0.78s]                                                                                                                                                                                                                                                   
Finished <<< rosbridge_msgs [6.05s]                                                                                                                                                                                                                                               
Finished <<< btcpp_ros2_interfaces [15.4s]                                                                                                                                                                                                                    
Starting >>> behaviortree_ros2
Finished <<< delib_ws_24_interfaces [19.7s]                                                                                                                                                                                                                
Finished <<< skiros2_msgs [34.0s]                                                                                                                                                                                                                           
Starting >>> skiros2_common
Finished <<< skiros2_common [0.96s]                                                                                                                                                                                                                                              
Starting >>> skiros2_world_model
Finished <<< rosapi_msgs [35.4s]                                                                                                                                                                                                                              
Finished <<< flexbe_msgs [35.5s]                                                                                                                                                                                                                         
Starting >>> flexbe_core
Finished <<< skiros2_world_model [1.19s]                                                                                                                                                                                                        
Starting >>> skiros2_skill
Finished <<< flexbe_core [0.95s]                                                                                                                                                                                                          
Starting >>> flexbe_testing
Starting >>> flexbe_mirror
Starting >>> flexbe_input
Starting >>> flexbe_webui
Finished <<< behaviortree_ros2 [22.3s]                                                                                                                                                                                                                                         
Starting >>> btcpp_ros2_samples
Finished <<< skiros2_skill [2.08s]                                                                                                                                                                                                                             
Starting >>> skiros2_task
Starting >>> skiros2_gui
Finished <<< flexbe_input [2.00s]                                                                                                                                                                                                                              
Finished <<< flexbe_mirror [2.30s]                                                                                                                                                                                                                                     
[... cut ...]
Finished <<< rosbridge_suite [0.42s]                                                                                                                            
Finished <<< pyrobosim_flexbe_btcpp [8.65s]                                                                                        
Finished <<< ros_bt_py_interfaces [1min 18s]                             
Starting >>> ros_bt_py_web_gui
Finished <<< ros_bt_py_web_gui [0.68s]                          
Starting >>> ros_bt_py
Finished <<< ros_bt_py [1.44s]                  

Summary: 49 packages finished [1min 21s]
ROS Distro: 	 jazzy
ROS Domain ID: 	 0
ROS Workspace: 	 /delib_ws

matthias-mayr, 2024-10-15:
Great. Feel free to re-open in case there's a similar issue.

closed 2024-10-15 by matthias-mayr